TrueOpenVR/TrueOpenVR-DIY #2: arduino head tracking not working

quirosaur, 2019-08-19:
I have followed the guide on crating head tracker with arduino nano and mpu 6050, i calibrated and flashed the correct code in arduino. in serial monitor, lot of stuff is showing up, but when selecting arduinocontrollers as a driver, only gray screen shows up. when using keyboard driver, vr screen works.

r57zone, 2019-08-19:
Use the Splitter driver (Keyboard HMD & Arduino controllers)

quirosaur, 2019-08-19:
if i'm using it, the screen is still gray

r57zone, 2019-08-19:
Steam by default turns off the screen if there is no head movement.
To track the head, you need to use the RazorIMU driver.

quirosaur, 2019-08-19:
razorIMU has same result, screen turns gray as soon as i start steamVR, bigscreen or vivecraft.

closed 2019-09-22 by r57zone