TrueOpenVR/TrueOpenVR-DIY #4: RazorIMU driver/MPU6050 rotation spasming on load

PTVoltz, 2020-01-02:
Hi! First off, a bit offtopic but I LOVE your program - so easy to use, and has meshed really well with my other stuff.

Now, onto the problem:
I recently looked into replacing my original Rotation Tracking setup with your RazorIMU driver, as it seems to work better. For reference, I'm using an Arduino Pro Micro (though it says "Leonardo" in the IDE), and an MPU6050. They're different from your example (which uses an Arduino Nano), and the pinout diagrams look quite different, but it seemed to work prefectly fine with the Calibration script so I went ahead with it.
However, after calibrating, installing the correct firmware, changing TrueOpenVR to RazorIMU, and launching SteamVR, the rotation system went Haywire - the whole thing seemed to be spinning and the screen was flashing between looking forward, up, rolling to the side, and various other things.

I'm presuming something isn't working due to the difference between your setup and mine, but I have no idea where to even begin diagnosing the problem due to being incredibly new to Arduino and I2C.
[Pinout diagram for my setup is here](http://edtracker.org.uk/images/edtracker/edtracker_circuit_diagram.jpg) - it was originally an EDTracker which I repurposed for full 360 tracking, but could never figure out the Maths to do the angles correctly.
If you can't reproduce the problem, I can send a video showing what's happening in SteamVR.

r57zone, 2020-01-02:
Hi, there may be a problem in the fact that the output in Serial is incorrect, there should be [3 float variables in binary](https://github.com/TrueOpenVR/TrueOpenVR-DIY/blob/master/HMD/Arduino/Arduino.Output.Bin.ino). The speed should be "115200". Also you need to check sample app "GetData" in main archive. In yaw, pitch and roll, values should be between ~ -180 to ~ 180.

PTVoltz, 2020-01-02:
Alright, thanks. First off: Speed is already set to 115200 - that's default used by everything else, so I just presumed that was correct.
Can't check the Get Data program right now, as I don't have enough HDD space to install Visual Studio. Will try to free up some space, but that might take a while.
Finally, when you say "Serial" you mean the Serial Monitor in the Arduino IDE, right? If so, then that's definitely the root of the problem: this is the gist of what I'm getting through the monitor (bear in mind this was run for like 30 seconds)
`⸮f⸮BN⸮C⸮	⸮a$⸮B⸮	C25	Ç⸮⸮⸮Y⸮⸮⸮vs-C⸮,m⸮AT⸮⸮,⸮-C⸮]W⸮⸮ԛ¿⸮A⸮⸮^⸮B⸮⸮B,dBB⸮S(⸮*7⸮⸮q(⸮⸮#⸮⸮½^Ci⸮[⸮⸮0⸮⸮⸮⸮-⸮⸮m⸮⸮⸮⸮#,⸮⸮,,A䀼⸮⸮CY$⸮A⸮^⸮⸮iSC⸮⸮A⸮⸮⸮⸮6CN⸮⸮⸮o⸮⸮B⸮⸮1C3],⸮KL⸮⸮ZB⸮G⸮Ags⸮⸮⸮<Q⸮Aqsr⸮⸮⸮⸮⸮Ŏ⸮A⸮⸮⸮/⸮k⸮⸮Ӭ⸮⸮⸮B⸮(⸮`
I can't post the whole thing because it's too long and Github dosn't like it, but you get the idea.

r57zone, 2020-01-02:
GetData sample app can be taken [here](https://github.com/TrueOpenVR/TrueOpenVR-Core/releases/download/0.7/TrueOpenVR.Samples.zip), from [releases](https://github.com/TrueOpenVR/TrueOpenVR-Core/releases), Visual Studio is not needed.
What firmware do you have on Arduino? The conclusion seems to be true, this is a binary output.

PTVoltz, 2020-01-02:
OK, so a bit of a silly from me: I was originally using an old version of the code (I think). I have since downloaded the GIT source (the TrueOpenVR-DIY main branch) and installed the HMD_MPU6050_DMP6.ino code in there. I no longer get the weird spinning/glitching, and it seems to be reading the data at least somewhat properly.

However, I am now getting a rather different problem, and no less annoying - the outputs seem to be mixing together.
Pitching the headset up will result in an output pitch down and yaw slightly to the left, turning the headset to the left (yaw left) will result in a pitch up and slight yaw left output, headset roll right results in a yaw left and slight pitch down - you get the idea. Also, after any movement the output will slowly decrease in size, until the view is straight forwards again.
The GetData sample (which I missed first time around, very sorry) shows the same - outputs vary wildly from supposed input, and slowly return to yaw/pitch/roll 0.

r57zone, 2020-01-02:
Need calibration, there is an instruction in the description. There are also some problems with inclines, but I don’t know how to fix x, it seems to be a firmware bug.

r57zone, 2020-01-02:
I use GY 85, there are no problems with the firmware and 360 tracking (RazorIMU firmware), but it seems that the new versions of this tracker are incompatible with the old firmware.

PTVoltz, 2020-01-02:
Oh, really? Missing calibration would mess with it that much? Alright, fair enough - should have figured that out on my own - thanks! Will try calibrating with the updated software and see if that helps.

r57zone, 2020-01-02:
https://youtu.be/sKuiGC6Mxf0?t=184

PTVoltz, 2020-01-02:
Yep, I know about that - my previous Arduino setup used the exact same calibration script, however from what I could tell all that did was correct for yaw drift.
And it seems I may have been somewhat right - calibrating the sensor made absolutely zero difference to th problem I'm having - all rotations go in the wrong directions, and mrge together in weird ways. I think I'll upload a video showing my problem, if you can hold on a minute or two...

PTVoltz, 2020-01-02:
OK, I wanted captions but editing takes a while - this text will have to do.
The video at the link is a direct recording of the SteamVR Compositor. While recording, I moved the headset quite slowly - first pitching up, returning to center, yaw to the left, yaw to the right, return to center, roll left and right, and then pitch down. As you can tell, it isn't quite working how it's supposed to...
Also, sorry about the Streamable link - Github doesn't support mp4, and this was the fastest thing I could find.
link: https://streamable.com/e1m87

r57zone, 2020-01-02:
Yes, I saw this problem, but I don’t know how to fix it. Probably the problem is in the firmware for MPU6050.

PTVoltz, 2020-01-02:
Ah, right. Welp, guess I'll stick with my original setup for now then. Thanks anyway!

r57zone, 2020-01-02:
Perhaps [this firmware](https://github.com/terminal29/Arduino-Tracker-Plugin) does not have this problem, but you need to replace the output from the quaternions with the angles yaw, pitch, role.

closed 2020-01-02 by r57zone