TrueOpenVR/TrueOpenVR-DIY #5: Add vibration to controllers

Artemiy-Z, 2020-11-11:
How can i add vibration?

r57zone, 2020-11-11:
Are you talking about Arduino controllers?

Artemiy-Z, 2020-11-11:
Yes

r57zone, 2020-11-11:
Have you assembled your controllers entirely? Do they work well and only vibration remains?
I don't know how exactly the vibration is done, but it seems like a mosfet transistor is needed to regulate the speed of rumble motor.

Artemiy-Z, 2020-11-11:
I haven't made controllers yet... But i want to know how get information about vibration from steamvr.

Artemiy-Z, 2020-11-11:
I plan to make it in the future

r57zone, 2020-11-11:
There is no vibration support for Arduino yet, because first you need someone to check to make vibration, in principle, on Arduino (motor + Mosfet), and only then add support.

Artemiy-Z, 2020-11-11:
https://techzeero.com/arduino-tutorials/vibration-motor-with-arduino/
How i can see, vibration motor can work with Arduino

Artemiy-Z, 2020-11-14:
https://valvesoftware.github.io/steamvr_unity_plugin/api/Valve.VR.ISteamVR_Action_Vibration.html
Can i/you use this to add vibration (if it possible on Arduino)

r57zone, 2020-11-14:
[Vibration support](https://github.com/TrueOpenVR/SteamVR-TrueOpenVR/blob/master/OpenVR/samples/driver_sample/driver_sample.cpp#L792) is included in the driver in SteamVR, but it is not added to the Arduino sketch and [TrueOpenVR Adrduino driver](https://github.com/TrueOpenVR/TrueOpenVR-Drivers/blob/master/C%2B%2B/ArduinoControllers/ArduinoControllers.cpp#L484).

Artemiy-Z, 2020-11-14:
Is it work?

closed 2020-12-01 by Artemiy-Z