TrueOpenVR/TrueOpenVR-DIY #7: Arduino controller needs more information and guide

Rmdhn1, 2021-12-08:
Firstly, it may be good to put in soldering / schematics on the wiring, and then it doesn't list how many of the parts, i assume you need 2 gyros as it mentions 2 imu, but it would be great if the specifics were mention and more images to guide, anyways, cheers as this is the only arduino controllers for steam vr that i can find

Rmdhn1, 2021-12-08:
another question is, can this be used using Driver4VR webcam system's

r57zone, 2021-12-08:
I don’t have a scheme for connecting two IMUs to one Arduino, I don’t know how to do it now, I only know that it’s possible, so I added support.

There is no compatibility with Driver4VR.

If you are looking for inexpensive good controllers, then you can get the Razer Hydra, which is the best solution for today.

closed 2021-12-08 by r57zone

Rmdhn1, 2021-12-08:
how about one IMU only? how will it be wired up?

Rmdhn1, 2021-12-08:
i presume i follow the normal wiring diagrams

r57zone, 2021-12-08:
Soldering scheme https://github.com/TrueOpenVR/TrueOpenVR-DIY/blob/master/HMD/HMD.md#rotation-tracker
Fimwares https://github.com/TrueOpenVR/TrueOpenVR-DIY/blob/master/Controllers/Controllers.md#arduino-controllers
Setup in TOVR https://youtu.be/bkB__YZT6cA