TrueOpenVR/TrueOpenVR-DIY #8: How to do positioning DIY Arduino Controller?

Katrin0522, 2022-03-28:
ENG:
I made an Arduino controller, since you did not write a diagram for connecting a second IMU to Arduino. Option 1 is not suitable, I thought about the second option, where PS EYE is needed. I read the instructions for PsMoveService - there is no description of my binding DIY Arduino Controller. If you know, tell me how.
RU:
Я сделал контроллер на основе Arduino, так как вы не написали схему подключения второго датчика - 1 Вариант не подходит. Я думал на счёт второго варианта, где нужен PS Eye. Почитал инструкцию по PsMoveService - там не описана привязка DIY контроллера как контроллер. Если в ы знаете как это осуществить - скажите пожалуйста.

r57zone, 2022-03-28:
https://github.com/r57zone/PSMoveService-receive-position/blob/master/README.md

Katrin0522, 2022-04-04:
Eng
> Thank you, only position tracking worked HMD. But when I try to use the controller, ps eye start tracking the controller like a HMD. HMD color Red, Controller is Blue... What am I doing wrong?
Rus
> Спасибо, трекинг позиционный работает на очках. Но когда я пытаюсь использовать контроллер, камера PS захватывает позицию контрлллера вместо шлема.. Цвет шлема красный, контрлллера - Синий. Что я делаю не так?

Katrin0522, 2022-04-04:
Or do I need to do something in TrueOpenVR to make everything work fine?

r57zone, 2022-04-04:
You need to change the index somewhere in the psmoveservice configs 

Нужно изменить index где-то в конфигах psmoveservice

Katrin0522, 2022-04-04:
Индекс в конфиге контроллера в конфиге psmove? На какое значение нужно изменить?

Katrin0522, 2022-04-04:
Я потыкал конфиг, теперь вообще контроллер перестал показываться. В TrueOpenVR стоят на контроллерах PSMoveService и ArduinoControllers. Никаких признаков жизни

r57zone, 2022-04-04:
Не помню, но там есть порядок какой-то в PSMoveService конфигах

Katrin0522, 2022-04-04:
Это конечно прекрасненько, извиняюсь, но можете конкретнее сказать что за порядок и в каких конфигах?

r57zone, 2022-04-04:
Я бы рад, но уже не помню, настраивал года 4 назад

Katrin0522, 2022-04-04:
Я пересмотрела множество видео, не нашла нужного, а в инструкциях тоже ничего особого нет...qwq
А я уже в один вложила ~ 1.5к, будет грустно если всё в пустую будет:)

r57zone, 2022-04-05:
virtual hmd на 1
https://youtu.be/IunGVk89-TY?t=182
hmd color
https://youtu.be/IunGVk89-TY?t=257

Katrin0522, 2022-04-05:
Удивительно, но это я смотрела и делала как там, трекинг головы идеален, а вот контроллер вместе с TrueOpenVR, крашит стим

r57zone, 2022-04-05:
Контроллер PSMove?

Katrin0522, 2022-04-05:
Контроллер на Arduino, отдельно без сервиса PSMove в выборе на positions в TrueOpenVR в SteamVR спокойно работает, крутится как положено, кнопки работают. 
А как только выбираю сервисы в опции позиция на контроллер, то при запуске стимвр идёт крашя посредством чего все дополнения отрубаются, и при повторном включении без изменения настроек происходит тоже самое.

Katrin0522, 2022-04-05:
У меня 64Бит система..  в каждом сплитере своей битности должны быть своей битности dll.. так? -3-

r57zone, 2022-04-05:
ага, например, PSMoveService64.dll

Katrin0522, 2022-04-05:
Интересный вопрос.. сейчас посмотрю

r57zone, 2022-04-05:
Есть еще такой вариант https://www.youtube.com/watch?v=bkB__YZT6cA можно попробовать

Katrin0522, 2022-04-05:
Посмотрела.. предлагаете изменить эти жва параметра ? могу попробовать завтра.. И вопрос.. за 4 года еще не появился способ подключения двух IMU одновременно?:)

r57zone, 2022-04-05:
Поддержка в драйвере есть, но поддержки в Arduino нет, проектом больше не занимаюсь

Katrin0522, 2022-04-05:
Жаль.. тогда попробую то что было в видео.. если это не удовлетворит, брошу идею создания бюджетного vr комплекта..

Katrin0522, 2022-04-05:
Не поленилась глянуть что сейчас у меня.. так понимаю я нужно будет поиграться с этими двумя парамнтрами.. shoulder yaw и pitch.. завтра попробую

Katrin0522, 2022-04-07:
В общем.. попробовала.. позиционирование конечно относительное.. но хоть что-то.. каким образом можно изменить длину предплечья и плеча и вообще высоту контроллера в драйвере? Ибо в vr у меня не состыкуются руки по примерному положениб, из-за чего мозг ломается и путаюсь.

r57zone, 2022-04-07:
В конфиге, в папке TrueOpenVRSettings

Katrin0522, 2022-04-09:
Спасибо за всю помощь... Попробовала я этот тип позиционирования, жить можно. Но тут такой прикол, что у меня исчезли краши с сервисом😅. Сейчас имею почти рабочее позиционирование(дрожит, но это не в этой теме обсуждать буду). То что запускается нормально драйвер через раз-два поняла. Но теперь образовалась новая проблемка. 1-ый контроллер собрала на оригинальном Arduino, 2-ой начала собирать этой ночью на китайской "рабочей" реплике Arduino - данные выводит с задержкой, да только не в оси вращения, а в форму Buttons и в форму что нажато. Пробовала переподключать платку, перезаливать скетч, тоже самое. Может это из-за китайской ардуинки?

r57zone, 2022-04-09:
Вроде нет разницы китайская или нет

Katrin0522, 2022-04-09:
По моему есть и велика в моем случае. оригинальная плата быстро работает и нормально и спокойно дает значения гироскопа куда надо, и драйвер сам подхватывает пк.
Вот драйвер на китай искала минут 20, как минимум, нашла фиг пойми где, загружает скетч на китай через раз, ну и симптомы которые я описала в прошлом коменте.. Буду покупать оригинальную ардуинку, думаю всё решится.

Katrin0522, 2022-04-09:
Такой вопрос, Nano и  Uno отсутствуют, есть Leonardo, подойдет для контроллера (с точки зрения возможностей)

Katrin0522, 2022-04-09:
М... Купила Leonardo.. залила скетч.. в COM порте пустота..

Katrin0522, 2022-04-10:
Спасибо за всю ту помощь, что вы дали.

closed 2022-04-10 by Katrin0522