TrueOpenVR/TrueOpenVR-DIY #9: controllers stand on back side.

osmangokc, 2022-03-28:
i am using [Controller_MPU6050_DMP6.ino](https://github.com/TrueOpenVR/TrueOpenVR-DIY/blob/master/Controllers/Arduino/Controller_MPU6050_DMP6.ino) code on my project. it works nearly perfect. But on steam vr i saw hand controllers on back side  (180 degree of home position).  i think; i should add +180 degree on startup. how can set the startup rotation ?

thanks a lot.

r57zone, 2022-03-28:
Hi, try resetting the rotation and it will center the controllers. Also calibrate the hmd.

osmangokc, 2022-03-28:
i am using Android mobile phone for head tracking with ivry plugin. I don't
know whether it has calibration option or not.

How can I reset the rotation? From code or screen?

i tried resetting center on ivry plugin but it didn't work.

29 Mar 2022 Sal 00:27 tarihinde Кладовая ***@***.***> şunu
yazdı:

r57zone, 2022-03-28:
https://github.com/TrueOpenVR/SteamVR-TrueOpenVR#centring

closed 2022-04-12 by r57zone