TrueOpenVR/TrueOpenVR-DIY #10: How to launch controllers from the first time?

Katrin0522, 2022-04-11:
Такая проблема, каждый раз когда включаю SteamVR, у меня почти всегда один из контроллеров не поворачивается, либо же нажимаются хаотично все кнопки что есть. А может и вообще не запуститься. Только с 15-20ой попытки возможно запустить нормально со всеми поворотами.. Как можно облегчить ситуацию/исправить? Использую два 3-ех метровых провода, на коротком проводе те же самые признаки. 10 Windows 1809.

Katrin0522, 2022-04-12:
Нажимать F11 будучи уже запущенным в SteamVR? или перед надо?

r57zone, 2022-04-12:
После запуска да, когда появится задержка

r57zone, 2022-04-12:
Пока невозможно, если есть возможность запишите видео с работой контроллеров, было бы очень интересно посмотреть. Сам стим вр там или в игре какой.

Katrin0522, 2022-04-12:
> Сделал еще очиску очереди на центрирование, нужно нажать на контроллера - "Grip" + "System", на обоих контроллерах.
> [ArduinoControllersNew2.zip](https://github.com/TrueOpenVR/TrueOpenVR-DIY/files/8474928/ArduinoControllersNew2.zip)
> Если есть возможность еще проверить, было бы здорово и я обновлю для всех это версию.
> 

Завтра на правый контрлллер сделаю кнопку System(а то я забыла про центрование).
Завтра как доделаю дело с контроллером - проверю и скажу что да как. Фичу с F11 ещё тестю, вроде работает..

Katrin0522, 2022-04-13:
Так.. поставила то что вы скинули, при сбросе правый контроллер сбрасывает в не те значения, изза чего косит налево по y на градусов ~ 45, изза чего приходится перезапускать стимвр чтобы все было ок, сейчас протесты прошлую версию жрайвера..

Katrin0522, 2022-04-14:
Проверила, сброс на кнопках работает не очень, либо правый либо левый контроллер сбрасывает с погрешностью поворота на 20-45 градусов влево по x. Сбрасываю через F8, так удобнее

r57zone, 2022-04-14:
Хорошо, спасибо за тесты, обновлю тогда для всех в ближайшее время.

closed 2022-04-14 by r57zone