TrueOpenVR/TrueOpenVR-DIY #12: [Feature request] Using Kinect for the movement tracking

Mineplayerminer, 2022-05-13:
I found out about this project just a few days ago and it's really interesting. I would like to make a VR headset myself, but I didn't have any PSMove cameras. Would it be possible to implement a webcam or an Xbox 360 Kinect support for the headset/controller movement? I hope this project goes on. 😄

r57zone, 2022-05-13:
Hi, unfortunately I don't have a kinect to check the work and test it. You can try driver4vr or other projects, it seems there are.

Mineplayerminer, 2022-05-14:
Driver4VR doesn't work for me. It's even paid and there's no longer a legacy version. Even if using the standalone installer. Would it be possible to support controllers like Joy Cons? I'm using PhoneVR as my headset, but, I just need some controllers. It would be nice to use them.

r57zone, 2022-05-14:
Maybe one day when I buy joycons