TrueOpenVR/TrueOpenVR-DIY #14: Arduino

LabrentFiero, 2022-11-27:
Здравствуйте. Можно как нибудь добавить в прошивку поддержку аналоговых пинов для поддержки джойстиков?

r57zone, 2022-11-27:
Привет, поддержка есть, можно изменять значение от 0 до 1 плавно, 0.1, 0.2 и так далее, если речь о триггерах, если о джойстиках, то также можно изменять, от -1 до 1, с 0 в центре. В скетче просто для упрощения показано как изменять. Насколько я помню там нужно использовать analogRead функцию и читать с аналогового пина, там кажется 1024 значение, так вот вычесть от туда 512 и разделить на 512, тогда будет от -1 до 1 значения.

r57zone, 2022-11-27:
Также конкретно в этом проекте Arduino джойстики работают так себе, рекомендую проект HadesVR, там контроллеры доделаны лучше https://github.com/HadesVR/HadesVR

LabrentFiero, 2022-11-28:
Спасибо. Дело в том, что я не могу сейчас использовать HadesVR, потому что я поторопился и купил mpu6050. В том проекте используется датчик с магнитометром. Я конечно их уже заказал, но пока будут ехать хотел бы что-то на вашем проекте сделать)

LabrentFiero, 2022-11-28:
В голову мысль пришла - а что если mpu9250 на контроллеры использовать, а mpu6050 зафиксировать на шлеме вместо плеча и использовать как датчик для позиционирования в пространстве для контроллеров?

r57zone, 2022-11-28:
Можно, если есть прошивка хорошая, с калибровкой, для mpu9250.

LabrentFiero, 2022-11-29:
Здравствуйте, есть ещё вопрос! Поставил драйвер на Razer Hydra. Работает хорошо, но где можно переназначить кнопки и изменить их свойства именно на программном уровне? Дело в том, что хват не срабатывает в некоторых играх.
[image] полагаю что эта область должна быть былая
[image]как здесь

r57zone, 2022-11-29:
Вот драйвер на Razer Hydra https://github.com/r57zone/Razer-Hydra-SteamVR-driver переназначения клавиш нет, но все работает как надо, грип на двух кнопках дублируется. В TrueOpenVR драйвер для гидр неисправный, частичная начальная реализация.

r57zone, 2022-11-29:
Не забудьте отключить контроллеры в TrueOpenVR, чтобы не было сбоев и конфликтов.

LabrentFiero, 2022-11-29:
Спасибо за помощь!

closed 2022-11-29 by r57zone