TrueOpenVR/TrueOpenVR-DIY #15: Problem with arduino controllers

Sussy123432, 2022-12-27:
So i made arduino controllers using arduino uno, connected all buttons, mpu 6050 flashed the code. I changed in true open vr sethings to arduino controllers. And nothing happens. I dont have any idea. Please help me.

r57zone, 2022-12-27:
Do you have any helmet connected?
Have you changed the com ports in the config file?
If the helmet is not connected, you need to make a splitter plugin, select emulation for the helmet and arduinocontrollers for the controllers in it.
You can also check the work in the GetData program.

Sussy123432, 2022-12-28:
What is splitter plugin and helmet?
I tried with GetData program but again, controllers are not working

r57zone, 2022-12-28:
[image]
[image]
[image]

Sussy123432, 2022-12-28:
Still not working 
[image]
and steam vr doesnt detect them 
[image]

r57zone, 2022-12-28:
Have you correctly specified the monitor resolution, com ports?
Have you configured your SteamVR config?

Sussy123432, 2022-12-28:
yeah, i tried with keyboard driver and it works perfectly. And i configured steam vr. Monitor solution is great, and com ports too

r57zone, 2022-12-28:
it means that the wrong Arduino sketch may be flashed, I don’t know, but it looks like the driver does not see the com ports, otherwise there would be controllers icons

Sussy123432, 2022-12-30:
wait, i tried now, and he showed me up the controllers on steam vr, but they are not working

Sussy123432, 2022-12-30:
i dont know why, but when i connect arduino controllers, they looks like connected, but any respond, even if i press button on them

karizr45, 2023-08-01:
I got error like this, is it solved?

karizr45, 2023-08-01:
bruh

karizr45, 2023-08-01:
if i will find a solution i will write here for everyone

karizr45, 2023-08-01:
i wrote email to owner and he is not responding now

SussyD3V, 2023-08-01:
give him time

SussyD3V, 2023-08-01:
> i wrote email to owner and he is not responding now

I tried to debug it but the exe is compiled. @r57zone if u can post decompiled code, I would look into it (:

karizr45, 2023-08-01:
> give him time

Ok

karizr45, 2023-08-01:
Wait I got that error with connected controllers but no respond but what if we just connect them to PSMoveService?

karizr45, 2023-08-01:
I will try this later

r57zone, 2023-08-01:
If the controllers appear, then everything is working. It remains to flash the correct Arduino sketch with a compatible output (Serial.write ...).
All sources are published, GetData sources is in TrueOpenVR-Samples

karizr45, 2023-08-02:
ok i tried to change SerialBT to Serial and it's not working now

karizr45, 2023-08-02:
Still not working but I flashed controller.ino file because sketch with mpu6050 is not flashing at the moment with error

karizr45, 2023-08-02:
OK what do I need to do for repairing controllers? @r57zone

karizr45, 2023-08-02:
nvm it's started working

karizr45, 2023-08-03:
no it's not