TrueOpenVR/TrueOpenVR-DIY #24: Контроллер вращается

karizr45, 2023-08-19:
Здравствуйте, я тут допилил вращение контроллера через библиотеку fastIMU но когда я его пытаюсь двигать он начинает вращатся в разные стороны просто так и это не соответствует коду помогите пожалуйста

r57zone, 2023-08-19:
Привет, не откалиброван может, сложно сказать, если yaw, pitch, roll не изменяется, то вращаться не будет. В FastIMU используется кватернион, а в этом проекте yaw, pitch, roll.

Как я писал много раз, контроллеры здесь работают криво, имеется замок Эйлера, поэтому хорошо контроллеры работать не будут, это начальная реализация, которую я врятле буду доделывать до конца. 

Если нужны дешевые контроллеры рекомендую RazerHydra, если нужно собрать на Arduino, то есть проект HadesVR, может однажды он добавит поддержку эмуляции позиционирования (упрощенное) на основе вращения.

karizr45, 2023-08-20:
Калибровать я пробовал все также, если FastIMU не сработает то как можно переделать все под библиотеку I2CDev для esp32?