TrueOpenVR/TrueOpenVR-DIY #25: Контроллеры не определяются

karizr45, 2023-09-21:
Привет, я нашел баг связанный с драйвером
короче я пытался протестировать контроллер все идет нормально кнопки нажимаются вращение есть но драйвер считает что они выключены и пишет: "controllers off" когда я на обоих поставил один и тот же порт
Помогите пожалуйста

r57zone, 2023-09-22:
Привет, в SteamVR если норм, то баг в GetData, может быть да

karizr45, 2023-09-22:
Он даже в стиме не определяется

karizr45, 2023-09-22:
я нажимаю кнопку в getData и они отображаются но пишет что нету контроллеров

karizr45, 2023-09-22:
сейчас я сменил тип драйвера со сплита на просто ардуино контроллеры и заработало

karizr45, 2023-09-22:
почему то с PSMoveService он не работает, есть идеи почему, может я чтото не сделал

r57zone, 2023-09-22:
Может баг в Splitter драйвере, а какой используется? Splitter просто или SplitterAdvance?

karizr45, 2023-09-22:
advance

r57zone, 2023-09-22:
Можно тогда простой проверить, может в нем бага нет

karizr45, 2023-09-23:
Может нету потому что я еще не ставил PSMoveService