TrueOpenVR/TrueOpenVR-DIY #26: и снова проблемка(я полюбому достал уже)

karizr45, 2023-09-30:
Привет, Вообщем проблема в том что у меня почему то данные поворота идут в триггер и джойстик вместо поворота, я не знаю с чем это связано потому что программа написана верно и все где и должно быть (а еще там данные увеличиваются постоянно)

r57zone, 2023-09-30:
Привет, кажется нужно сбрасывать через ALT + F7 чтоли, не помню уже, в инструкции к драйверу должно быть написано. Как я где-то писал уже запариваться не стоит, драйвер работает плохо, даже если получится сделать там есть замок эйлера и прочее, вращение будет плохим.

karizr45, 2023-09-30:
понял спасибо

r57zone, 2023-09-30:
F5 надо нажать, должно помочь https://github.com/TrueOpenVR/TrueOpenVR-Drivers/blob/master/C%2B%2B/ArduinoControllers/ReadMe.txt

karizr45, 2023-09-30:
ок спасибо

karizr45, 2023-09-30:
не помогает он просто убирает контроллеры с экрана и все

karizr45, 2023-09-30:
есть идеи как заменить этот драйвер но важно чтобы работало по сериалу?

karizr45, 2023-09-30:
HadesVR явно не подойдет этот проект был лучше всего подходящим

r57zone, 2023-09-30:
Чтобы сделать хороший проект у меня недостаточно знаний, поэтому самый простой доступный вариант это купить Razer Hydra контроллеры